seehase/neowx-material #11: N/A instead of date and time in both current archive reports

copain60, 2022-11-28:
There seems to be a small error in archive, which exists already in the original Neoground repo.
On current year and current month reports ONLY, always N/A is shown below the site title instead of current date and time.
All other archive reports are correct.

W0CHP, 2022-11-29:
Known issue. Not sure why this happens.

Here's my workaround on [my site](https://wx.w0chp.net):

in [header.inc](https://github.com/seehase/neowx-material/blob/df364da3948a0d17faa7858c9a9167d6b3b0e991/src/header.inc#L20):

```python
<strong>$station.location</strong><br>
#if $active_nav is 'archive'
<span class="font-small"></span>
#else 
<span class="font-small">$current.dateTime</span>
```

If you have a better solution, feel free to submit a Pull Request.

seehausen, 2022-11-29:
I like your aproach @W0CHP
the date field does not make sense on archive page because it is already shown
[image]

I'll fix this

seehausen, 2022-11-29:
Please not merge yet, there is still a bug on the month and year pages... I'll work on it

seehase, 2022-11-30:
Changes are merged, issue can be closed

closed 2022-11-30 by seehase