seehase/neowx-material #16: Min and Max Values in History

Pogs2004, 2022-12-06:
I’m trying to get my head around the min and max values  listed in the history section of skin.conf. Why are there min and max values for each table, when only min, max and average values are shown as a single line? I have the same min and max values for all three temp stats but always seem to have cells with no colour:

www.360shetland.co.uk/weather/history.html

There seems to be no response to the minimum values, is it just the max values that are used? For instance all my minimum temps below 0 are the same colour, when the min values line goes down to -6.

seehase, 2023-01-21:
It is always a tripple (min, max, color)
e.g.
```
[[uv_max]]
        obs_type = UV
        aggregate_type = max
        minvalues =        0 ,         2 ,        4 ,        6 ,        8 ,       10 ,       12 ,       15
    	maxvalues =        2 ,         4 ,        6 ,        8 ,       10 ,       12 ,       15 ,       50
    	colours =   "#04EC97", "#05EF3D2", "#2BF207", "#8AF408", "#E9F70A", "#F9A90B", "#FC4D0D", "#FF0F2D"
```
that means value beween 0..2 will use color #04EC97
2..4 -> #04EC97

so min and max defines upper and lower bound of the interval
did that answer your question?

Pogs2004, 2023-01-21:
Yes, but as far as I can see the minvalues are ignored, only the max values are used. My maxvalue scale used to start at 0 but everything below 0 was the same colour. I have changed the maxvalue scale to start at -2.9ºC and it works perfectly:

obs_type = outTemp
        aggregate_type = max
        minvalues = -5, -4, -3, -2, -1.5, -1, 0, 1.5, 2.5, 5, 10
        maxvalues = -2.9, -1, 0, 3, 6, 9, 12, 15, 18, 21, 24
        colours =   '#3C58CE', '#0186E7', '#02E3EA', '#04EC97', '#05EF3D', '#2BF207', '#8AF408', '#E9F70A', '#F9A90B', '#FC4D0D', '#FF0F2D'

seehase, 2023-01-21:
OK, I see at least some problems

1. ranges should not overlap
2. there should be no gaps
3. you always need vaild tripples

your config:
```
minvalues =        -5,      *-4*,        -3,        -2,       1.5,        -1,         0,       1.5,       2.5,         5,    10
maxvalues =    *-2.9*,        -1,         0,         3,         6,         9,        12,        15,        18,        21,    24
  colours = '#3C58CE', '#0186E7', '#02E3EA', '#04EC97', '#05EF3D', '#2BF207', '#8AF408', '#E9F70A', '#F9A90B', '#FC4D0D', *missing* 
```
I think what you want to achieve should look like this

```
minvalues =        -5,      -2,9,        -1,         0,         3,         6,         9,        12,        15,        18,       21
maxvalues =      -2.9,        -1,         0,         3,         6,         9,        12,        15,        18,        21,       24
  colours = '#3C58CE', '#0186E7', '#02E3EA', '#04EC97', '#05EF3D', '#2BF207', '#8AF408', '#E9F70A', '#F9A90B', '#FC4D0D', '#ff0000'   
```

seehase, 2023-01-22:
each red box is handled as a tripple
[image]
Do you still see a bug? or can we close the issue?

Pogs2004, 2023-01-22:
OK, I've finally got it! I was reading rows rather than columns, so the min, max and colour are a single group. Your tabbed example above was helpful! I've spent the afternoon sorting my history config, looking good now. Thanks for the help!

https://www.360shetland.co.uk/weather/history.html

closed 2023-01-22 by seehase