seehase/neowx-material #17: Corrupt Formatting on history.html

Pogs2004, 2022-12-14:
Corruption of tables in history.html (see screenshot). I've noticed this twice now, the last time was at 00:05 on 01/12/2022 when a new month was added, then again at some time last night when there were changes to some of the December stats.

[image]

It seems to sort itself out after a while, it certainly did on 01/12/2022. I have made no changes to the history.html.tmpl or skin.conf in the last few days, and the historygenerator.py hasn't been touched or changed.

Pogs2004, 2022-12-15:
I have had to remove the max windspeed from my stats as it still remains broken. We had heavy snow on Monday which affected my ultrasonic anemometer, it recorded a gust of 146mph! I think this may be causing an issue with it, not sure.

seehausen, 2022-12-16:
I will have a look this weekend
Thanks for reporting

Pogs2004, 2023-01-03:
Happy New Year! I removed the erroneous wind gust from the database and now it's formatting correctly, however it is a concern that a three figure gust (which is highly possible up here) breaks the formatting, when there are plenty three figure values in the rainfall section.

I still get temperature values that are not coloured, I have played around with the min and max values but can't seem to find a range that works for all temperature values. https://www.360shetland.co.uk/weather/history.html

seehase, 2023-01-21:
Hi @Pogs2004 
I think your problem could be a wrong color value `#05EF3D2` it has 7 digit but must be 6
[image]

Pogs2004, 2023-01-21:
Well spotted! That’s it fixed now :)

seehase, 2023-01-21:
According to the wrong formatting
how many vallues did you correct in the DB and in which table, was it just one value, can you reproduce it.
I changed some values but did not get the same error

Pogs2004, 2023-01-21:
It was a maximum windspeed value, IIRC it was around 200mph. I only changed a single record and that fixed it. I’ll have a look in the SQLite history to see which record it was and change it to a high value to see if it breaks it again.

seehase, 2023-01-21:
did you changed the value in table "archive" or "archive_day_wind" or?
maybe we can reproduce and fix it

Pogs2004, 2023-01-21:
I removed it from the archive table then rebuilt the daily summaries for that day

seehase, 2023-01-21:
hmmm... I changed max wind speed for Jan. 2022
https://solar.seehausen.org/weewx/history.html
but chart is still not broken... maybe convertion from km/h -> mph? not sure
[image]

If it happens again, we can have a look before you change the database

Pogs2004, 2023-01-23:
When I was playing around with the min, max and colour values yesterday I broke the formatting in the same way, so I think this issue is directly related to #16 with overlaps in the min and max values. I will keep an eye on it but I will close this issue now as resolved.

closed 2023-01-23 by Pogs2004