seehase/neowx-material #19: Remove Minimum Values from UV and Solar Radiation Cards

Pogs2004, 2023-01-07:
The UV and Solar Radiation cards show current, minimum and maximum values, however the minimum values are irrelevant as they will always be zero. Is there a quick fix to remove these values?

[image]

W0CHP, 2023-01-07:
Yes. And in fact, I omit the "lo" UV value on my own site. You need card logic like this in your template(s) immediately after the card title `</h3>` tag:

```python
        #if $name == "UV"

        <div class="row">
          <div class="col-3 text-muted font-small lo-text">
          </div>
          <div class="col-6">
            <h4 class='h2-responsive '>$current.UV</h4>
          </div>
          <div class="col-3 text-muted font-small hi-text">
            $day.UV.max <br>
            ($day.UV.maxtime.format($Extras.Formatting.datetime_today))
          </div>
        </div>

        #else

        <div class="row">
          <div class="col-3 text-muted font-small lo-text">
            $getVar('day.' + name + '.min') <br>
            ($getVar('day.' + name + '.mintime').format($Extras.Formatting.datetime_today))
          </div>
          <div class="col-6">
            <h4 class="h2-responsive">$getVar('current.' + name)</h4>
          </div>
          <div class="col-3 text-muted font-small hi-text">
            $getVar('day.' + name + '.max') <br>
            ($getVar('day.' + name + '.maxtime').format($Extras.Formatting.datetime_today))
          </div>
        </div>

        #end if
```

Notice that the `low-text` class div is simply empty when the card name called is "UV".

Pro tip, you can use logic like this anywhere in your cheetah templates.

Pogs2004, 2023-01-08:
> Yes. And in fact, I omit the "lo" UV and Radiation values on my own site. You need card logic like this in your template(s) immediately after the card title `</h3>` tag:
> 
> ```python
>         #if $name == "UV"
> 
>         <div class="row">
>           <div class="col-3 text-muted font-small lo-text">
>           </div>
>           <div class="col-6">
>             <h4 class='h2-responsive '>$current.UV</h4>
>           </div>
>           <div class="col-3 text-muted font-small hi-text">
>             $day.UV.max <br>
>             ($day.UV.maxtime.format($Extras.Formatting.datetime_today))
>           </div>
>         </div>
> 
>         #else
> 
>         <div class="row">
>           <div class="col-3 text-muted font-small lo-text">
>             $getVar('day.' + name + '.min') <br>
>             ($getVar('day.' + name + '.mintime').format($Extras.Formatting.datetime_today))
>           </div>
>           <div class="col-6">
>             <h4 class="h2-responsive">$getVar('current.' + name)</h4>
>           </div>
>           <div class="col-3 text-muted font-small hi-text">
>             $getVar('day.' + name + '.max') <br>
>             ($getVar('day.' + name + '.maxtime').format($Extras.Formatting.datetime_today))
>           </div>
>         </div>
> 
>         #end if
> ```
> 
> Notice that the `low-text` class div is simply empty when the card name called is "UV".
> 
> Pro tip, you can use logic like this anywhere in your cheetah templates.

Thanks @W0CHP that's sorted it on the index, I'll work through the rest of the templates. It's little things like this that make a good weather station front end stand out from the crowd!