seehase/neowx-material #20: Add Days with Rain Count

Pogs2004, 2023-01-23:
I thought this would be a simple bit of code but it’s turning out to be more complex! It would be a good feature to add the number of days with rain to the week, month and year page rain cards, and in the month / year reports. 

You can see an example here (showing 0 as it’s not currently working): https://www.360shetland.co.uk/weather/month.html

I installed the XAggs extension from here: https://github.com/tkeffer/weewx-xaggs

It has an aggregate function that looked like it would do the job. I used $month.rain.avg_ge((‘0.2’, ‘mm’, ‘group_rain’)) to return the number of days with rain equal or over 0.2mm, however it doesn’t appear to work. Is there a simple way to display the number of days with rain in the given archive period, $week, $month and $year?

W0CHP, 2023-01-24:
@Pogs2004 Please file an issue with Tom, the author of the XAggs extension: https://github.com/tkeffer/weewx-xaggs/issues

closed 2023-01-24 by W0CHP

Pogs2004, 2023-01-24:
I was hoping this could be done without using an extension, hence the reason I posted it as a possible enhancement.

W0CHP, 2023-01-24:
It will definitely require a new search extension and tags.