seehase/neowx-material #21: Station/System uptime displays in seconds and Cloudbase not charting

glenosAU, 2023-03-29:
Hi,
Its likely these are simple issues I do not understand and not related to the code.

I have updated to the latest version of neowx-material from here and found the station and system uptimes are being displayed in seconds, can they be displayed in days, hours, minutes is this a change in the skin or footer.inc?

Also since I changed to neowx-material last year I have had a calculated cloudbase tab displayed but it does not chart. See http://users.on.net/~milikare/weather/yesterday.html. There is data in the database stored as `archive_day_cloudbase`. This is possibly something simple as well.

Thanks
Glen

glenosAU, 2023-03-29:
I have solved the Cloudbase issue by reconfiguring the database. The schema did not have cloudbase in the archive table but has a daily summary for cloudbase by reconfiguring I have corrected the database.

W0CHP, 2023-03-31:
> This is due to a breaking change in WeeWX 4.10

Good catch...will commit shortly. Thank you.

closed 2023-03-31 by W0CHP

glenosAU, 2023-03-31:
Thanks everyone.