seehase/neowx-material #24: Months names in English instead of my language

stalkerGH, 2023-07-24:
Hello seehase.

Thank you for your improvement of NeoWX skin for Weewx! I've updated my Weewx installation, translated string in history.html.tmpl but months names abbreviations are still generated in English. How/where can I change them?

W0CHP, 2023-07-24:
1. Have you deleted the rendered HTML file and re-generated the reports?
2. Month/day, etc. names for WeeWX are actually derived from the host/server operating system, not WeeWX itself; and certainly not the skin. So you will need to change the locale/i18n setting to your locale on the server itself, if it is not already,

stalkerGH, 2023-07-24:
1. Yes, I've removed all files from /var/www/html/weewx. Didn't re-generated reports because Weewx does it on every run (I think so).
2. I got my locale set, it's visible in Weewx log (OS name, Python version etc.).

stalkerGH, 2023-07-24:
OK, I got the solution. First I have done things you recommend - stopped Weewx, cleared WWW directory, re-created reports, checked locale settings (they were perfectly correct - `locale -k LC_TIME` has proven that month names are translated), finally started Weewx again. But it failed, month names are still in English.

So I looked into the skin's code and found it - in skin.conf :) There is a section named [HistoryReport] with `monthnames` directive. I had to translate it - 30 seconds of work - and everything is working as expected.

:)

stalkerGH, 2023-07-24:
Thanks for help. Problem is now solved so I close the issue.

closed 2023-07-24 by stalkerGH

W0CHP, 2023-07-24:
@stalkerGH awesome! Nice work. Cheers 🥂