seehase/neowx-material #28: Additional tables in History

sapiens007, 2024-06-15:
Hi all,

how can I add some more tables in the history.html, such as warm_days (> 25 °C), hot_days (> 30°C), desert_days (> 35 °C)?

I searched and added in skin.conf and history-template, but it doesn't work.
I think I need a change on more places...

Maybe somebody has an idea?

Thanks a lot!

W0CHP, 2024-06-15:
You would likely need to add them as tags/definitions in the history generator:
<https://github.com/seehase/neowx-material/blob/master/bin/user/historygenerator.py>

> Hi all,
> 
> how can I add some more tables in the history.html, such as warm_days (> 25 °C), hot_days (> 30°C), desert_days (> 35 °C)?
> 
> I searched and added in skin.conf and history-template, but it doesn't work. I think I need a change on more places...
> 
> Maybe somebody has an idea?
> 
> Thanks a lot!

sapiens007, 2024-06-15:
Thank you very much - this was the missing link!

sapiens007, 2024-06-15:
Problem solved, was a user error...

closed 2024-06-15 by sapiens007