seehase/neowx-material #35: How to add more stats?

stalkerGH, 2024-12-31:
I'd like to add more stats, for example: max/min amplitude of temperature during a day (24 h), numbers of day with temperature or rain sum not higher or not lower than x and so on. How can I do that?

seehausen, 2025-01-12:
could be done possibly, but I'm not really an expert
if the data is available in the database, it should be strait forward to add additional cards, but there i would need more information

stalkerGH, 2025-01-16:
Thank you for the reply. I'll ask then on Weewx mailing list.

seehausen, 2025-01-16:
If the data is available (in weewx) we can adjust the skin

seehausen, 2025-02-23:
@BliXem1 
I haven't tried, but if it is available in WEEWX it should be able to add to the theme

If you have the "lightning" defined in your Database and weewx, you should add the metric "lightning" to the list I would say

```
    [[Appearance]]

        # The order of values cards (left column)
        values_order = outTemp, outHumidity, barometer, windSpeed, rain, snowDepth, dewpoint, windchill, heatindex, inTemp, inHumidity, UV, ET, radiation, appTemp, cloudbase, extraTemp1, extraHumid1, extraTemp2, extraHumid2, extraTemp3, extraHumid3, extraTemp4, extraHumid4, extraTemp5, extraHumid5, extraTemp6, extraHumid6, extraTemp7, extraHumid7, extraTemp8, extraHumid8

        # The order of chart cards (right column)
        charts_order = outTemp, windchill, barometer, rain, snowDepth, windSpeed, windDir, windvec, UV, ET, radiation, outHumidity, inTemp, inHumidity, appTemp, cloudbase, extraTemp1, extraHumid1, extraTemp2, extraHumid2, extraTemp3, extraHumid3, extraTemp4, extraHumid4, extraTemp5, extraHumid5, extraTemp6, extraHumid6, extraTemp7, extraHumid7, extraTemp8, extraHumid8
```

W0CHP, 2025-03-02:
> Is there a way to add a lightning detector (stats)?

A chart and a "card" would need to be created in order to leverage and display lightning data. There currently are no provisions for that in this fork.

seehausen, 2025-03-03:
@W0CHP OK now I got it
Is there already a fork that contains this? otherwise I can add it
But I need someone who can test it
What metrics should be added and how to display the charts (area, bar, radar)

BliXem1, 2025-03-06:
I hope this will added soon then :)

seehausen, 2025-03-12:
Added basic charts for lightning values.
I do not have any test data to check, so can someone have a look? or can provide some test data for me.
We can probably build a nice combined chart from the data

Support is desired

closed 2025-03-12 by seehausen