seehase/neowx-material #41: skin.conf hi/lo_value_color not working

andrewjmcginnis, 2025-01-13:
Maybe I'm thinking this is something it's not but-

Within `skin.conf` exists two values for high and low values' color on the cards (with my uncommented lines):

```
        # Here you can set a hex code for the text color of low / high values
        # on all cards. By default they are grey. (quotes are needed on values)
        # lo_value_color = "#03a9f4"
        # hi_value_color = "#f44336"
        lo_value_color = "#03A9F4"
        hi_value_color = "#F44336"
```

within `head.inc` is this block:

```
## Appearance settings

#if $Extras.Appearance.lo_value_color != ""
<style>
    .lo-text {
        color: $Extras.Appearance.lo_value_color !important;
    }
</style>
#end if

#if $Extras.Appearance.hi_value_color != ""
<style>
    .hi-text {
        color: $Extras.Appearance.hi_value_color !important;
    }
</style>
#end if
```

I've tried double quotes, single quotes, upper- and lower-case, with and without #, restarted the weewx service, deleted everything within `/var/www/html/weewx/`, but my cards' hi/lo values are always grey (the default).  

[image]

Where else should I be looking to get this color setting to apply?

iiseppi, 2025-01-13:
Don't use capitals in hex code, and then it works.

andrewjmcginnis, 2025-01-13:
I finally played with it enough to discover the problem.
[image]

Here's the `head.inc` relevant sections where the custom color should be passed and applied (but isn't)

```
#if $Extras.Appearance.lo_value_color != ""
<style>
    .lo-text {
        color: $Extras.Appearance.lo_value_color !important;
    }
</style>
#end if

#if $Extras.Appearance.hi_value_color != ""
<style>
    .hi-text {
        color: $Extras.Appearance.hi_value_color !important;
    }
</style>
#end if
```

the color variable isn't passed with the "#" character. I added # to `color: $Extras.Appearance.hi_value_color !important;` and `color: $Extras.Appearance.lo_value_color !important;`

so now it's 
```
#if $Extras.Appearance.lo_value_color != ""
<style>
    .lo-text {
        color: #$Extras.Appearance.lo_value_color !important;
    }
</style>
#end if

#if $Extras.Appearance.hi_value_color != ""
<style>
    .hi-text {
        color: #$Extras.Appearance.hi_value_color !important;
    }
</style>
#end if
```

and the colors work:
[image]

iiseppi, 2025-01-14:
Interesting. I have to check my head.inc file, because I just edited the skin.conf file just like you show’d on this topic, except using non capital letters in hex code. 
Then I just restarted weewx and now they appear as color coded.

seehausen, 2025-01-14:
Do we have an issue?
for me it also works with using 
```
        lo_value_color = "#03a9f4"
        hi_value_color = "#f44336"
```
or should we adjust the header.inc

andrewjmcginnis, 2025-01-14:
I'm not sure where the issue ultimately lies.  Case of the hex code doesn't matter I've found. 
I tried having double ## in skin.conf (to see if it would strip just one #), but that ## was getting passed through to the rendered html.
This no-color issue was happening with the OG neowx-material, on both a weewx 4.10.2 (python v3.11.2) and 5.1.0 (python v3.9.2) install.  I've only installed your fork on the 4.10.2 install, and that's where I did my digging.  After figuring out the need for a # outside of the $Extras.appearance.ho/hi_value_color in head.inc, I did the same on my 5.1.0+OG neowx-material server, and it worked as well.

seehausen, 2025-01-14:
But why don't I see this problem?
I just have the code from my fork without any changes, and works out of the box

I can check what happens if I add the # in the HTML and not in the variable.
if that works I can change the code

just for my understanding
your skin.conf looks like this (with "#")
```
        lo_value_color = "#03A9F4"
        hi_value_color = "#F44336"
```

and the template has also an additional "#"

```
<style>
    .lo-text {
        color: #$Extras.Appearance.lo_value_color !important;
    }
</style>
```
from my perspective does not look right, but if it solves the problem

andrewjmcginnis, 2025-01-14:
This is my skin.conf section; not using a # in the color, because it was getting stripped. Not that _first character_ is stripped, but the # is being removed because of some formatting(?) bug treating the # as some kind of special character in the html generator. I could put it back in, as the commented lines show it, but it doesn't survive.
```
        # Here you can set a hex code for the text color of low / high values
        # on all cards. By default they are grey. (quotes are needed on values)
        # lo_value_color = "#03a9f4"
        # hi_value_color = "#f44336"
        lo_value_color = "03a9f4"
        hi_value_color = "f44336"
```

the head.inc, as I have it:
```
#if $Extras.Appearance.lo_value_color != ""
<style>
    .lo-text {
        color: #$Extras.Appearance.lo_value_color !important;
    }
</style>
#end if
```

seehausen, 2025-01-14:
@andrewjmcginnis 
can you check PR https://github.com/seehase/neowx-material/pull/46

andrewjmcginnis, 2025-01-14:
> @andrewjmcginnis can you check PR #46 also old implementation worked for me, but if it is more reliable we can use your version

save a stupid person from himself.... am I just looking at the changes and saying "yeah, that looks right", or...? Because those changes to `skin.conf` and `head.inc` do match my own modifications.

closed 2025-01-14 by seehausen