seehase/neowx-material #49: Problem after upgrade

dg9bja, 2025-02-05:
Hi,
thanks for your work on this skin. Yesterday I did upgrade my system from 4.10.2 to 5.1 and after a good start reporting needs a lot of CPU and memory. It doesn't finish after waiting 20 minutes.
I did try different things like change database schema and did a database reconfigure. Read about calculated values that could be the reason but don't know how to find it.
Or does the generator need that much time because all values since 2018 are being used for first time?

Debug =1 did not throw anything special.

dg9bja, 2025-02-06:
Solved my problems. Maybe performance of mysql ist too bad with a lot of data (since 2018 every minute). Changed to sqlite and solved my problems. Needed some time but running now.

closed 2025-02-06 by dg9bja

seehausen, 2025-02-06:
OK, good to know
for me migration from 4 -> 5 did not worked well, I installed everything from scratch..