seehase/neowx-material #56: Added chart card, but plot is not generated

vpreatoni, 2025-03-07:
Hi,
I added a chart (_altimeter_), adding it to the `chars_order` option, but plot is not generated.
Value card is shown properly, and the _altimeter_ data exists into the database, but plot is not generated.

This is how it looks:

[image]

W0CHP, 2025-03-07:
Have you defined the chart in the templates as well as the chart type? We can’t see your code so we won’t know.

image.png (view on web)

vpreatoni, 2025-03-07:
No, sorry. I think you are talking about editing the `index.html.tmpl` file, but unfortunately I don't know how to do HTML coding.

seehausen, 2025-03-07:
I will check and provide a first draft, then you can check

seehausen, 2025-03-07:
hi @vpreatoni 

I added missing code for altimeter in this branch
https://github.com/seehase/neowx-material/tree/feature/new_cards
can you check that it works as expected?

You can see my results here
https://weewx.seehausen.org/

are more values missing, or should I also add other values?
Holger

vpreatoni, 2025-03-10:
Thanks! that worked

[image]

> hi [@vpreatoni](https://github.com/vpreatoni)
>
> I added missing code for altimeter in this branch https://github.com/seehase/neowx-material/tree/feature/new_cards can you check that it works as [...]

seehausen, 2025-03-12:
Should be fixed, please use latest version 1.38.0

closed 2025-03-12 by seehausen