seehase/neowx-material #60: Add some way to indicate station is OFFLINE

vpreatoni, 2025-03-25:
When weewx dies, main page will show latest data, but there is no quick visual way to realize it is OLD data. I mean, you have to carefully check on the dates to realize system went offline or weewx crashed.

Eg.: this is how main page looks when weewx has been dead by a day or more:

[image]

There should be a customizable big GREEN/RED label on some place that indicates if station is online or offline based on last data received. The logic of that label should be embedded in HTML code, so if weewx dies, HTML can still show the label and change it's color accordingly.

seehausen, 2025-03-25:
Hi @vpreatoni 

Not sure If I can do that. 
If weewx dies, no new pages are created, so you only see the latest created and uploaded HTML pages.

One thing we could do is checking it the date on top left corner is not not older then xxx hours, and change color or do an alert... but this is just cosmetic.

[image]

BTW: I also have the problem that weewx main loop crashes because auf my used weatherstation driver.
My solution is to check every 3min via cronjob, that weewx is still running, and if not, restart weewx

for that I use the script "weewx_check.sh"

```
#!/bin/bash

dt=$(date '+%F %T');

#echo "$dt"  "Start weewx check"

STATUS="$(systemctl show -p SubState --value weewx)"

if [ "${STATUS}" = "running" ]; then
   # echo "$dt" "OK weewx service up and running"
  dt=$(date '+%F %T');
else
   echo "$dt" "not running ... restart weewx service"
   # /etc/init.d/weewx restart
  systemctl restart weewx
fi
```
and cronjob
```
# m h  dom mon dow   command
*/3	 *	 *	 *	 *	/var/myTools/weewx_check.sh weewx 2>&1 >> /var/log/weewx_check.log
```
that works good for me...
If you want I can add the script to the repo or just create the script, make it exceutable and call it via cron

seehausen, 2025-03-25:
I will integrate some script, with a config of maxDIfHours in skin.conf, to be able to configure behavior or to switch it off

and yes I also have the service restart on failure... but still I saw some cases, where this does not work, so I have both
systemd + script

uktricky, 2025-03-25:
Might be a workaround in the meantime but I run a script on my system - Sends me a telegram message to my phone if last update in database is > 10mins old - Has served me well across a number of systems 

check-weewx-db.sh
LOG="/home/user/logs/check-weewx-db.log"
echo $LOG
DB="/var/tmp/weewx.log"

echo "PI4-WEEWX" > $LOG
echo "================================" >> $LOG

# Grab Last update time from weewx database
docker exec -i rt-mariadb mariadb -u <dbuser> -p<dbpassword> <db> -e "select datetime,FROM_UNIXTIME(datetime) from archive order by datetime
desc limit 1;" > $DB

# Get database update time from LogFile
lastdb=$(more $DB | grep - | cut -b 12-30)
lastdbepoch=$(date -d "$lastdb" +%s)

# What are we dealing with in terms of Values
echo "Last DB Update : " $lastdb            >> $LOG
echo "DB Epoch       : " $lastdbepoch       >> $LOG

# Get current date/time but take 10 minutes off this
clock=$(date --date="now - 10 minutes")
clockepoch=$(date --date="now - 10 minutes" +%s)

echo "Clock Drifted  : " $clock              >> $LOG
echo "Clock Epoch    : " $clockepoch         >> $LOG

# Work out how long since the last update or when I should have seen one based on sub 10 minutes
timedrift=$(( $lastdbepoch - $clockepoch))
echo "Time Drift     : " $timedrift          >> $LOG

maxdrift=600;		# Maximum allowed drift in seconds
echo "Max Drift      : " $maxdrift           >> $LOG

# Check that the database has been updated in the last 10 minutes
if [ "$timedrift" -gt "$maxdrift" ]; then
  echo "No Recent WeeWX DB Update"           >> $LOG
  # Send a telegram message
  $HOME/scripts/telegram.sh "$(cat $LOG)"
else
  echo "All is well"                         >> $LOG
  cat $LOG
fi

seehausen, 2025-03-26:
@uktricky 
Nice script, you get informed if something is wrong. but additionally i would still do a restart automatically if main loop crashes

seehausen, 2025-03-26:
@vpreatoni 
I created a first draft to show the labels
please check out this branch
https://github.com/seehase/neowx-material/tree/feature/offline_label
(only 3 files needs to be changed)
* header.ind
* js.inc
* skin.conf

How it works you can see at https://weewx.seehausen.org/index.html, if pages are older than 4 minutes it shows "OFFLINE"

and have a look in the PR https://github.com/seehase/neowx-material/pull/61 there I raised some questions.

vpreatoni, 2025-03-26:
Hi and thanks!!

I did some changes to js.inc file, since your version was not working on my system.
I had to add the UTC parameter to moment, and use now.diff function for getting minutes difference.
Also added an indication if there is a date parsing error.
This is the time format I use here: **"DD/MM/YYYY HH:mm UTC"**

```

if (lastUpdate) {
            const lastUpdateDate = moment.utc(lastUpdate.textContent, "$Extras.Header.current_date_pattern");
            const now = moment.utc();
            const diffMinutes = now.diff(lastUpdateDate, "minutes");
            const statusLabel = document.getElementById("status-label");

            if (!lastUpdateDate.isValid()) {
                statusLabel.textContent = "DATE ERROR";
            }
            else if (diffMinutes > $Extras.Header.offline_threshold_minutes) {
                statusLabel.textContent = "OFFLINE";
            } else {
                statusLabel.textContent = "ONLINE";
            }
        }

```

> [@vpreatoni](https://github.com/vpreatoni) I created a first draft to show the labels please check out this branch https://github.com/seehase/neowx-material/tree/feature/offline_label (only 3 files [...]

seehausen, 2025-03-26:
@vpreatoni 
I found a smarter solution without converting dates
please have a look again, we have to check if it works with UTC as well

If you have a public site, please send me the link to your page to seehausen@gmx.de (or post it here)

If you want to debug, just use the development tools of your browser
e.g. Chrome on MacOS: cmd + opt + i

[image]

closed 2025-03-27 by vpreatoni