seehase/neowx-material #67: Wrong encoding of HTML tags in tooltips

stalkerGH, 2025-05-07:
I'd like to ask how to properly use HTML tags in tooltips. In my language .conf file I declared:

[Units]

    [[Labels]]

	meter		  = " m", " m"
        day               = " dzień",    "<sup>d</sup>"
        hour              = " godzina",   "<sup>h</sup>"
        minute            = " minuta", "<sup>m</sup>"
        second            = " sekunda", "<sup>s</sup>"

because I like time units as superscript. But when I point the mouse cursor to trend arrow, I see it like this:

[image]

Is it some magic method to avoid it? Markdown? Other variable?

seehausen, 2025-05-07:
I can have a look, probably we need to implement "something"
do we need more tags than <sub>, or do we should support all html tags?
and this should only apply on tooltip but not on other places where the units is used

stalkerGH, 2025-05-07:
In the almanac it renders properly:

[image]

And I use such code in `almanac.html.tmpl`:

```
<div class="col">
<h6 class="$Extras.color-text"><i class="wi wi-day-sunny"></i>
     $gettext("daylight")</h6>
     $almanac.sun.visible.long_form(format_string="%(hour)d%(hour_label)s %(minute)d%(minute_label)s %(scond)d%(second_label)s")<br/>($sun_visible_change.long_form(format_string="%(minute)d%(minute_label)s %(second)d%(second_label)s") $change_str)
</div>
```

seehausen, 2025-05-10:
Looks good

[image]

I will create a Pullrequest for you to test

seehausen, 2025-05-10:
@stalkerGH 
please check this branch
https://github.com/seehase/neowx-material/tree/feature/support_html_tooltip
it contains the fix for your tooltips

stalkerGH, 2025-05-12:
> [@stalkerGH](https://github.com/stalkerGH) please check this branch https://github.com/seehase/neowx-material/tree/feature/support_html_tooltip it contains the fix for your tooltips

Sorry for the late reply, I was away for a few days.

Your fix works, the superscript is displayed as expected. Thank you very much!

stalkerGH, 2025-05-12:
> [@stalkerGH](https://github.com/stalkerGH) It looks that you use a polish language file do you want to provide your language file to the public? then let's us add this to the current version too
> 
> Just send me the pl.conf or create a PR containing the file

Of course I can upload the pl.conf file, but I would prefer not to create a new PR. First - I don't know how to do it (I don't want to mess something up, and I didn't make changes to my installation using Git), second - in Polish (and probably many other Slavic language too

[pl.conf.txt](https://github.com/user-attachments/files/20162239/pl.conf.txt)

s) the matter is unfortunately more complicated due to different endings of the same adjective. In English it looks simple:

highest temperature
highest pressure
highest rainfall

And in Polish it looks like this (look at the last letter of the adjective):

najwyższ**a** temperatura
najwyższ**e** ciśnienie
najwyższ**y** opad deszczu

So to make it look good, I had to change not only the language file, but also the history.html.tmpl file.

However, I prepared a fairly universal pl.conf file in which I use the terms "maximum" and "minimum". As a first approximation, this will work.

seehausen, 2025-05-12:
OK, then send me the file if you are ready.

can you check the implementation for HTML Tooltips?
so that I can merge it and add "pressure" in the next run... actually it is already done, just need to test

[image]

So let's do both together

seehausen, 2025-05-12:
I will add pl.conf as well to this PR
--> v 1.42.1

seehausen, 2025-05-12:
solved with https://github.com/seehase/neowx-material/pull/69

closed 2025-05-12 by seehausen