seehase/neowx-material #90: Missing building zip file

tfilo, 2025-08-21:
Hello, in original repository there is build process which creates ZIP file with skin and only required files. In this repository it was removed. I am aware that it is possible to install it by calling.

weectl extension install https://github.com/seehase/neowx-material/archive/refs/heads/master.zip

which will download whole repository as zip file. But it contains even files that are not needed for installing skin to weewx like all scss files, and few others. If someone changes scss file, he must than run build locally to create required css files. It would be more ideal to have github action which will run build and than create zip file with just required files on each release.

tfilo, 2025-08-21:
I have github action to create this zip file working in my own fork of original repository, i can make required modification and prepare pull request if you are interested in creating zip release with just required files and remove css files from repository and create them while build process.

seehausen, 2025-08-21:
We can provide a zip, if you like

I know that this
    weectl extension install https://github.com/seehase/neowx-material/archive/refs/heads/master.zip

contains more than needed, but thought this is "good enough" 
but yes we can provide a "setup.zip" per release