seehase/neowx-material #95: Help with changing language

tfilo, 2025-08-22:
Hello, while testing I would like to test slovak locale as well, not sure how to change locale. According to weewx documentation for this new translations format, I set in weewx.conf 

    [[Test]]
        skin = neowx-material
        enable = true
        HTML_ROOT = /var/www/html/
        lang = sk

and a tried this too:

    [[Test]]
        skin = neowx-material
        enable = true
        HTML_ROOT = /var/www/html/
        lang = sk_SK.utf8

and after restart, in both cases it still generates only in english. Original neowx had language set in skin.conf which worked fine for me. Not sure if this is some bug o I am doing something wrong. Can you point me to right direction how to make it work ? Thanks.

tfilo, 2025-08-22:
I wanted to try different language if it will work, and i discovered in DE localization you have multiple duplicit keys about battery status, it cause weewx to crash with DE localization.

seehausen, 2025-08-22:
I thought I have removed duplicates entries

to change language, 
The language of this skin is definde in weewx.conf file

```
   [[Defaults]]
        
        # Which language to use for all reports. Not all skins support all
        # languages.  You can override this for individual reports.
        lang = en
```

and/or

```
   [[NeowxMaterial]]
        # The neowx-material uses the 'neowx-material' skin
        skin = neowx-material
        lang = en
        enable = true
```

but I haven't test it with 
`[[Test]]`

I will check

closed 2025-08-22 by seehausen