seehase/neowx-material #113: Autorefresh

tfilo, 2025-08-25:
Now with autorefresh it takes me to top of page after refresh. What do you think about adding some javascript to header to save current scroll postion to sessionStorage on page unload and then restore scroll postion on page load? Not sure yet how it will work but i can try to implement it. What do you think?

seehausen, 2025-08-25:
Actually I don't care ;-) I like the auto-refresh, scrolling position is not so important... if you can fix it, please do so

BliXem1, 2025-08-25:
@tfilo is your website public accessable? If so, what is the link? Want to check it out :D

BliXem1, 2025-08-25:
Nice, I thought you had a lightning sensor, but it was only for test I guess? :)