seehase/neowx-material #137: Lightning strike counter stats not working

BliXem1, 2025-09-09:
Hi,

I've running Weewx with MQTT subscriber and I notice that there are 8 lightning counts. On the left it's working, but on the right, no lightning stats are showing anything. Screenshot:

[image]

Lightning:

            [[[[lightning_count]]]]
                ignore = False
                name = lightning_strike_count
                contains_total = true

            [[[[lightning_distance_km]]]]
                ignore = False
                name = lightning_distance

Any idea what is going wrong?

BliXem1, 2025-09-09:
I think it's because this:

`            contains_total = true`

tfilo, 2025-09-09:
Best is to check in database in archive table. It there is recorded distance but in lightning count there is null, than it is deffinitely issue with driver you use to record data to weewx.