seehase/neowx-material #148: Third arrow

BliXem1, 2025-09-25:
"Maybe we can also add a third arrow. We have one for up and down but nothing if the tempature (example) is stable."

Not urgent at all, so take your time to find out if it's possible :)

seehausen, 2025-09-25:
@BliXem1 
Good catch... the symbol was already implemented bug with a bug

should be fixed now

[image]

threshold for up or down arrow is now changes greater then 0.5

closed 2025-09-25 by seehausen

BliXem1, 2025-09-25:
Nicely done! How can I fix this? Just wait untill the new update?

seehausen, 2025-09-25:
just take latest code from
https://github.com/seehase/neowx-material
actually only the index.html.tmpl has been changed